botletics/Reflowduino #1: Include documentation for adding ESP32 functionality and standalone displays

vogelfreiheit, 2017-11-18:
I think adding some documentation for adding ESP32 modules and standalone displays would be great. There should be enough IO pins left for both, AFAIK.

botletics, 2017-11-20:
Hi there, thanks for the suggestion! Sorry I didn't see this earlier. Adding support for display shields is currently in the [to-do list](https://github.com/botletics/Reflowduino/wiki/To-Do-List) and ESP32 support sounds fun! I'll keep you updated when that happens. In the meantime I'm launching an Indiegogo campaign for Reflowduino on December 1st!

VinnyCordeiro, 2017-11-20:
There was a project of a reflow oven called [Reflow Château](https://hackaday.com/2015/02/02/reflow-chateau/) that appeared in 2015 on Hackaday that uses a Teensy 3.1 and a graphic touchscreen display. The original website of the project is down but there is a copy of it here on Github. I've made [a fork of it](https://github.com/VinnyCordeiro/ReflowChateau), as there are some missing files in the original repo.

The controller works but uses obsolete/discontinued components, namely the thermocouple IC and the solid-state relays. I had the idea to update the project, but never had the time to really do that. I hope that this project will help to improve yours. The original project uses the Apache License, so it's open source.

botletics, 2017-11-20:
Awesome, thanks! I'll check it out when I get a chance.

vogelfreiheit, 2017-12-16:
I might be able to work on this soon, just got a Xmas sized lot of components and boards that are too much of a PITA to do without a reflow oven.

botletics, 2017-12-19:
I'm working on making an Instructables tutorial and YouTube video about how to build a BLE Android app for the ESP32. Stay tuned! Maybe I could also build a custom PCB that uses the ESP32 dev board as the controller that snaps into some headers?!?!

tedder, 2018-01-09:
esp32? ooo. I've been modding my 'duino and figure I'll switch to an esp32, perhaps a wemos-with-OLED one. hmm.

botletics, 2018-01-09:
Haha yea that would have to be a completely separate board but it's doable. Just a matter of when I can find time for such things  :)

tedder, 2018-01-09:
Right, it'll replace the 32u4. No reason to interface with it. I'm just glad to see others may be interested in it.

botletics, 2018-03-24:
Just added ESP32 support! I made a "backpack" module for this [DOIT ESP32 dev board](https://www.amazon.com/Docreate-Development-Bluetooth-Ultra-Low-Consumption/dp/B0754DT83W/ref=sr_1_3?ie=UTF8&qid=1521924528&sr=8-3&keywords=esp32).

closed 2018-03-24 by botletics