botletics/Reflowduino #3: Use of a Halogen Floodlight as Reflow Oven

Reaper2u, 2018-08-11:
Hello botletics

I am using the ESP32 Version to try to drive a Halogen Floodlight as a mini Reflow Oven. Most important point for me is, that I can easily take it with me, should I need to.

My problem with that approach is, that it is way too fast. I basically get no soak phase, as the Halogen Floodlight heats up so quick. The whole process until it reaches the peak temperature takes about 30 seconds, which is way too short.

Is there a chance You can test that, and maybe give me a hint what I can do to actually have useable soak and reflow phases?

Thanks in advance.

Best regards

Reaper2u

botletics, 2018-08-11:
Sorry, I don't have the time and I am not interested in testing that. Best of luck with your project though.

closed 2018-08-11 by botletics

tedder, 2018-08-16:
@Reaper2u, what you need to do is tune [the pid parameters](https://github.com/botletics/Reflowduino/blob/master/Example%20Code/Reflowduino_Demo/Reflowduino_Demo.ino#L91) to account for the rate difference. I built a new board and wrote code from scratch to control my reflowduino (okay, so, not much left of it at that point). [Here are some good overviews of pid and how to tune it.](https://pinboard.in/u:tedder42/t:pid/)