botletics/Reflowduino #6: Reflowduino32: Relay not being triggered

Jizanthapus, 2019-07-19:
I've assembled the Reflowduino32 and Sidekick module and connected them to my toaster oven. However, when I start a reflow sequence the toaster oven does not receive power. 

I modified the simple blink program to test whether the Sidekick and ESP32 backpack modules were working:
```
#define relay 13

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  pinMode(relay, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  digitalWrite(relay, HIGH);
  delay(3000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  digitalWrite(relay, LOW);
  delay(3000);                       // wait for a second
}
```

This code works fine and I can see my oven come on and go off with the built in LED. 

When I flash back to the Reflowduino_ESP32_Demo code, nothing happens with the oven. The Bluetooth connection and thermocouple reporting work fine. While measuring pin 13 with a voltmeter, I do not see it go high while the process is running. 

It's possible that I've overlooked something very simple, but I can't think of anything off the top of my head. 

Thanks

botletics, 2019-07-19:
Did you start the reflow process with the Android app via Bluetooth?

Jizanthapus, 2019-07-19:
Yes. I hit "Start" and the blue LED came on but the oven did not.

botletics, 2019-07-20:
The oven will only turn on when the *red* LED on the Reflowduino board turns on. The blue LED when solidly lit means it's connected via Bluetooth. Perhaps I'll have to look into the app. In the serial monitor do you see the command from the app's START button being sent to the ESP32?

Jizanthapus, 2019-07-20:
When I plug the ESP in, I get a red LED and serial output showing the temperature. After hitting START on the Android app, the blue LED comes on and I see serial output stating that the reflow process has begun. I also get the reported temperature displayed and graphed in the app. No relay activation. After hitting STOP in the app, I the blue LED turns off and I get serial output stating that the reflow process has been aborted. 

Today though I had a slight bug with the Android app. When I hit the start button, it said that there was a runtime error in the app and it closed. Restarting the app allowed me to continue as normal. As it doesn't happen every time I hit START, I don't think this is related to the relay not functioning, but I thought I'd mention it anyway.

I hooked the backpack's power and SPI pins up to the ESP and routed pin 13 solely to an o-scope. I never saw that pin flicker when hitting START on the Android app.

botletics, 2019-07-20:
Pin 13 is the red LED. Pin 7 controls the relay per [this wiki page](https://github.com/botletics/Reflowduino/wiki/Pinouts). I'll look into it more when I get a chance, thanks for the heads up.

Jizanthapus, 2019-07-20:
I'm using the Reflowduino32 backpack for the ESP32 board. According to the [schematic](https://github.com/botletics/Reflowduino/blob/master/Schematics/Reflowduino32%20Backpack%20Schematic%20v1.png) and [code,](https://github.com/botletics/Reflowduino/blob/master/Example%20Code/Reflowduino_ESP32_Demo/Reflowduino_ESP32_Demo.ino#L60) pin 13 drives the MOSFET that activates the SSR controlling the oven.

botletics, 2019-07-20:
Oh sorry, I was looking at the wrong thing, thank you.

botletics, 2019-07-20:
Could you please screenshot what you see in the serial monitor? Thanks.

Jizanthapus, 2019-07-21:
Haha, no worries

I did a little more testing today...

It appears that line [316](https://github.com/botletics/Reflowduino/blob/master/Example%20Code/Reflowduino_ESP32_Demo/Reflowduino_ESP32_Demo.ino#L316) is where you are turning on the relay. So I added a little serial output there:
```
if (output > millis() - windowStartTime) {
      digitalWrite(relay, HIGH); // If HIGH turns on the relay
      Serial.println("Activated Relay");
      }
```

After starting the reflow process from the app, I never saw that output. I do get the `<-- ***Reflow process started!` from line [164](https://github.com/botletics/Reflowduino/blob/master/Example%20Code/Reflowduino_ESP32_Demo/Reflowduino_ESP32_Demo.ino#L164).

Jizanthapus, 2019-07-21:
[Here is the serial output](https://gist.github.com/Jizanthapus/48d9e5a6e959edf510303ae55729ed68) from my modified code

botletics, 2019-07-22:
So it looks like the app and Bluetooth is working but the relay isn't turning on? Weird. What are your settings for T_preheat, T_soak, and T_reflow on lines 81-83 in the Reflowduino_ESP32_Demo?

Jizanthapus, 2019-07-22:
Yes, yes, and I haven't changed them from the ones you provide (I am planning on using a low-temp lead-free solder paste).

The only modification to your code that I've made is inserting the serial output where the relay is supposed to go HIGH.

botletics, 2019-07-22:
So the only thing I can think of that could be the culprit is that I divided the preheat_rate, soak_rate, reflow_rate, and cool_rate variables by 1000 in lines 263, 276, 288, and 304 because of [this issue](https://github.com/botletics/Reflowduino/issues/5) but I haven't tested that yet. Could you try deleting the "/ 1000" part and see if it at least runs?

botletics, 2019-07-22:
Also, if you try the [PID_Test](https://github.com/botletics/Reflowduino/blob/master/Example%20Code/PID_Test/PID_Test.ino) it should turn the oven on.

Jizanthapus, 2019-07-23:
I tried PID_Test (with some [modifications](https://gist.github.com/Jizanthapus/fa0ab96277080cf2d81ee36eaa1868aa) for the ESP32) and removing the "/ 1000" in the Demo code. Both resulted in Pin 13 going HIGH on the START command. 

I think you are correct about the culprit.

botletics, 2019-07-23:
Hmmm OK. What if you change the " / 1000" to " / 1000.0" to make sure it's doing the math properly?

Jizanthapus, 2019-07-23:
[image]

It works!
Does that profile look about right to you?

botletics, 2019-07-23:
Ehhh it looks a little like a camel but it's approximately right. Maybe the PID needs some tuning. Did it work when you added the "1000.0" decimal place?

Jizanthapus, 2019-07-24:
Yep. That did the trick.

Based on the wiki and my results, I'm thinking maybe I need to tweak my derivative gain to counter some of that overshoot. Does that sound reasonable or am I misunderstanding the PID parameters?

Since it's working now, I'll go ahead and close the issue. Thanks for all your help!

closed 2019-07-24 by Jizanthapus

botletics, 2019-07-24:
Great, thanks for letting me know.

I think you should lower Kp and increase Kd a little but not too much. Protip: What you could do for testing is lower the preheat/soak/reflow temperatures so you don't have to wait for it to heat all the way up every time.